rai-opensource/spot_ros2 #612: Error in builiding spot_hardware_control: bondcpp is not provided

TuXun1999, 2025-03-06:
Hi, I am having an issue building up spot_ros2 locally. It seems that my system is missing the dependency bondcpp

[image]

I have installed  libbondcpp-dev but new issues have occurred: 

[image]

I suspect it's because the library is too old, so I tried building bondcpp from source. However, the guidance from the official github repo of bond (https://github.com/ros/bond_core) is quite unclear and I don't know what to do. Is there any idea how to solve issue? Could someone share how to build bondcpp for ROS2 and therefore my spot_ros2 can be built successfully? Thanks a lot for any help!

tcappellari-bdai, 2025-03-06:
https://github.com/bdaiinstitute/spot_ros2/pull/615#pullrequestreview-2665198026 should fix this once it gets merged

khughes-bdai, 2025-03-06:
Merged now, sorry about that! You can now rerun the install script or you can also get around it manually with `sudo apt install ros-humble-bondcpp`

closed 2025-03-06 by khughes-bdai

TuXun1999, 2025-03-07:
Two additional steps I did to run spot_ros2 locally: 
1. Install ros-humble-bondpy as well
2. Install aiortc and multipledispatch using pip 
Fortunately, it's working fine now. Thank you very much for the clarification.