rai-opensource/spot_ros2 #658: TF Tree problem when using SLAM Toolbox

Mans0Leo, 2025-04-28:
Hello,

I am a researcher at CRIStAL and we have recently bought a Spot. I am currently trying to use SLAM with it but it seems that the TF tree of the Spot is not compatible with SLAM toolbox 

The SLAM wants to have the link `map -> odom -> base_link -> vision`, meaning he wants to create `map -> odom`. But because the TF tree of the spot is already doing `body -> odom`, when the SLAM create `map -> odom`, it breaks `body -> odom`. I have tried to use different TF as `odom` and `base_link` frame but because `base_link` is always before `odom`, I cannot find a way to make it works.
My TF tree looks like [this](https://github.com/user-attachments/files/19936686/TF_Tree_With_Slam_Toolbox.pdf) with Slam Toolbox and like [this](https://github.com/user-attachments/files/19936927/TF_Tree_Without_Slam_Toolbox.pdf) without.

Do I have to mess with the URDF of the spot ? Or is there simply something I am missing ?

khughes-bdai, 2025-04-29:
Hey @Mans0Leo , What happens if you try to set `tf_root` as `odom` in your [config file](https://github.com/bdaiinstitute/spot_ros2/blob/main/spot_driver/config/spot_ros_example.yaml)? Then the TF tree will have `spot/odom` -> `spot/body`, and slam toolbox can publish `map` -> `spot/odom`?

Mans0Leo, 2025-04-30:
I cannot believe I did not check this. Yes it works thank you very much. I'll close the issue.

closed 2025-04-30 by Mans0Leo

PengzhaoWu, 2025-08-04:
@Mans0Leo Hi, I'm also trying to do SLAM with the spot, but encountered a lot of issues. I wanted to know if you have finished the SLAM. And if so, I also wanted to ask you some questions.

Mans0Leo, 2025-08-12:
@PengzhaoWu Hello, I was able to do SLAM with the SPOT. I also encountered a lot of issues and I am still working on getting better results. I'll gladly help you if I can.

For informations, I tried to use the camera already on the robot to do the SLAM but the result was poor. I probably could've improve the result but I decided to use a Velodyne 16 attached to the SPOT (Much like the Core I/O) to have better measures.

What is your issue ?

PengzhaoWu, 2025-08-12:
@Mans0Leo @khughes-bdai Hi, I basically achieve SLAM with Spot using Velodyne. But I found two major issues. First, I found there's a large  time mismatch between the timestamp of lidar and the timestamp of odometry. I have tried some methods to solve this issue, but all basically failed---map overlaps in RVIZ2 when performing SLAM. Therefore, given this situation, I switched to use only lidar date to do SLAM. However, the average rate of lidar data is very low. So, the position of robot on the map largely lags behind the position in the real world, which makes it less useful in practice. Would you guys have some suggestions on these issues for me?