rai-opensource/spot_ros2 #662: Can actions sent to the API be canceled or preempted?

kabirkedia, 2025-05-14:
We're exploring ways to improve our high-level planner and were wondering:
Is it possible to cancel or interrupt an action after it has been sent to the Boston Dynamics API?

For example, if a new higher-priority task comes in or if the planner determines that the current action is no longer relevant, can we programmatically stop or override that in-progress action?

Any guidance or best practices would be appreciated. Thanks!

heuristicus, 2025-05-16:
You should be able to do this by calling the `stop` service. I believe that will stop anything that's currently being executed.

khughes-bdai, 2025-05-16:
+1 to the `stop` service. As another alternative, if you are executing a robot command as an action, you can request cancellation of the ROS action and this will under the hood call `stop` on the command: https://github.com/bdaiinstitute/spot_ros2/blob/d1aef0ec668e80a8890bfac5437fc876571e8463/spot_driver/spot_driver/spot_ros2.py#L2253
If you try to send a robot command action while another robot command action is currently in progress, the currently executing action will be cancelled and the latest request command will execute on robot.

closed 2025-05-16 by khughes-bdai