rai-opensource/spot_ros2 #669: Error while attempting to use EAP 2 Lidar

JustinBoverhof, 2025-05-21:
From looking around the repo and other past issues, it seems like EAP 2 support is a newer addition to this repo. I am currently running the spot ros2 wrapper inside the provided docker container on my laptop. I have been able to successfully visualize pretty much all other sensor data except for the lidar. The topic does not seem to be getting any messages and when i run 

```
ros2 launch spot_driver spot_driver.launch.py config_file:=/ros_ws/src/robot.yaml launch_rviz:=True publish_point_clouds:=True
```

I get the following error 

```
[spot_ros2-1] [_channel.py:247] Exception in callback <function FutureWrapper.add_done_callback.<locals>.<lambda> at 0x7f795c196170>: AttributeError("'google._upb._message.RepeatedCompositeContainer' object has no attribute 'point_cloud'")
```
Here is my config file as well

```
# This is an example configuration file for Spot.  It is not used!

/**:
  ros__parameters:
    # Spot Login Information
    username: "my user"
    password: "my password"
    hostname: "my hostname"

    # Status Updates from Spot
    metrics_rate: 0.04
    lease_rate: 1.0
    async_tasks_rate: 10.0

    # Some boolean parameters
    auto_claim: True
    auto_power_on: False
    auto_stand: True

    # Estop Parameters
    estop_timeout: 9.0
    start_estop: False

    preferred_odom_frame: "odom" # Allowed values are: [odom, vision], without a prefix. This will be used in the odometry topic. https://dev.bostondynamics.com/docs/concepts/geometry_and_frames.html?highlight=frame#frames-in-the-spot-robot-world for more info.
    tf_root: "odom" # Allowed values are: [odom, vision, body], without a prefix. This will set the root frame in your TF tree.

    # These parameters may be overridden by the 'spot_name' and 'tf_prefix' launch arguments.
    spot_name: ""
    # frame_prefix: "" # Set an explicit prefix for all frames in your TF tree (an empty string is a valid prefix option).

    cmd_duration: 0.25 # The duration of cmd_vel commands. Increase this if spot stutters when publishing cmd_vel.
    rgb_cameras: True  # Set to False if your robot has greyscale cameras -- otherwise you won't receive data.
    initialize_spot_cam: False # Set to True if you are connecting to a SpotCam payload module.

    use_velodyne: True # Set to True to stream point clouds from the velodyne on Spot EAP
    velodyne_rate: 10.0 # The rate in Hz that the velodyne will stream at

    # You can uncomment and edit the list below if you only want to publish data from a certain set of cameras.
    # cameras_used: ["frontleft", "frontright", "left", "right", "back", "hand"]

    # The following parameters are used in the image stitcher node and were determined through a lot of manual tuning.
    # They can be adjusted if the stitched image looks incorrect on your robot.

    # Virtual camera intrinsic matrix [fx, 0, cx, 0, fy, cy, 0, 0, 1]
    # fx stretches the image left-right, fy zooms in, cx moves the image left-right, cy moves the image up-down.
    virtual_camera_intrinsics: [385.0, 0.0, 315.0, 0.0, 385.0, 844.0, 0.0, 0.0, 1.0]
    # Plane that the stitched image is projected on with respect to the virtual camera frame. 
    virtual_camera_projection_plane: [-0.15916, 0.0, 0.987253]
    # The distance from the virtual camera frame to the projection plane
    virtual_camera_plane_distance: 0.5
    # The stitched image will be of size (<frontleft image width>, <frontleft image height> + row_padding)
    stitched_image_row_padding: 1182

    # Change to True if missing gripper on arm
    gripperless: False

    publish_point_clouds: True
```

I don't think it is an issue with the lidar itself because features like autowalk using lidar and human/moving object detection appear to be working.

I will acknowledge that I am newer to using spot and if this is a simple issue to fix I apologize for wasting your time.

khughes-bdai, 2025-05-29:
Pinging @nhewitt99 as I know he has gotten this to work in the past!

nhewitt99, 2025-05-29:
Happy to take a look. I'm on some travel this week but should have a chance to try to replicate on Monday or Tuesday. This doesn't sound familiar from my experience.

kahleeeb3, 2025-06-03:
> I was able to get the lidar visualization working yesterday, but it took making some small edits to a driver file and a few of the wrapper files. When I have the time I will try to make a pull request. I would still be curious to see if other people are having issues with the current code or if it is something with my lab's robot.

I am also having this issue.

JustinBoverhof, 2025-06-03:
> I am also having this issue.

@kahleeeb3 if you have the time feel free to pull my fork of the driver and see if the lidar works for you. 

https://github.com/JustinBoverhof/spot_ros2/tree/lidar_fix

kahleeeb3, 2025-06-03:
> > > I was able to get the lidar visualization working yesterday, but it took making some small edits to a driver file and a few of the wrapper files. When I have the time I will try to make a pull request. I would still be curious to see if other people are having issues with the current code or if it is something with my lab's robot.
> > 
> > 
> > I am also having this issue.
> 
> [@kahleeeb3](https://github.com/kahleeeb3) if you have the time feel free to pull my fork of the driver and see if the lidar fix works for you.
> 
> https://github.com/JustinBoverhof/spot_ros2/tree/lidar_fix

@JustinBoverhof That was able to fix my issue. Thank you very much for the help. Instead of pulling your fork, I just made the same changes since it was essentially just replacing all of these lines in the `spot_wrapper`
```python
#callback_future.add_done_callback(self._callback)
callback_future.add_done_callback(lambda future: self._callback(future.result()))
```
and then replacing this line in `spot_driver/spot_ros2.py` file
```python
#local_time = self.spot_wrapper.robotToLocalTime(data.point_cloud.source.acquisition_time)
local_time = self.spot_wrapper.robotToLocalTime(point_cloud_proto.point_cloud.source.acquisition_time)
```
For context: I'm also using the EAP 2 and a docker container.

nhewitt99, 2025-06-04:
Sorry, still haven't had a good chance to replicate this myself. But I did double-check that I was using the clean Dockerfile back when I tested the addition of this feature last month and didn't need to make modifications like these. Glad that this sounds like a fix! Although I'm curious what could have introduced an issue like this between May 5 and now (potentially alongside the 5.0.0 release). The most recent commits in `spot_ros2`, `spot_wrapper`, and `spot-sdk` don't seem to have an obvious link to this.

PR #675 (2025-06-13): lidar data handling fix
Change to the driver to fix #669 to be used in tandem with the fix to the wrapper [here](https://github.com/bdaiinstitute/spot_wrapper/pull/169)
closed 2025-06-13 by khughes-bdai