rai-opensource/spot_ros2 #681: Unsupported image format in spot_image_publisher_node

TiBr-IGP, 2025-06-11:
Hello everyone,

After successfully installing the spot_ros2 package and spot_driver, I would now like to read the point clouds from the installed depth cameras, especially the _Velodyne VLP-16_ LIDAR. Unfortunately, there seems to be a problem with the image format of the sensors.

## Setup
Spot Explorer with Arm and Core (legacy)
Host PC with ROS 2 Humble connected via spot's access point

## Launch
The error occurs after starting the driver: 
$ ros2 launch spot_driver spot_driver.launch.py config:=spot_ws/src/cfg.yaml
Compared to the example file ‘spot_ros_example.yaml’, only the parameters ‘use_velodyne’ and ‘publish_point_clouds’ were changed to ‘True’ in cfg.yaml.

## Error
After starting the launch script, the following error appears repeatedly on the terminal:
**[spot_image_publisher_node-7] [ERROR] [1749653611.346499683] [image_publisher]: Failed to get images: Failed to get images: 6(STATUS_UNSUPPORTED_PIXEL_FORMAT_REQUESTED): ImageResponse ::bosdyn::common::Status unsuccessful**
This seems to result in the publication of any depth data failing.

## Unsuccessful attempts to resolve
After some research, I found the already closed [issue #555](https://github.com/bdaiinstitute/spot_ros2/issues/555), which suggested setting the parameter ‘rgb_cameras:=False’. Unfortunately, this did not work, nor did the accepted solution from #555, which involved installing an older version of Numpy (< 2) on the host PC.

## Questions
Is this error known? Are there any other possible solutions?

Kind regards

khughes-bdai, 2025-06-13:
Hi @TiBr-IGP ,

The `spot_image_publisher_node` that is displaying the error is getting data from Spot's onboard cameras. The `rgb_cameras` param should be set to false if your robot has greyscale cameras -- is that the case? And what version of numpy are you using? Seems like the solution from #555 was to upgrade to a > 2 version of numpy. 

The velodyne point cloud data is coming from the `spot_ros2` node so it shouldn't be affected by this error, on the topic `velodyne/points`. Do you see anything on there? There was also a recent bug fix for velodyne data publishing [here](https://github.com/bdaiinstitute/spot_wrapper/pull/169) and [here](https://github.com/bdaiinstitute/spot_ros2/pull/675) that just merged, so make sure you are on the latest main!

TiBr-IGP, 2025-06-17:
Hi @khughes-bdai,
the rgb_cameras param is indeed set to false. My current version of numpy is 1.24.1 since @mschweig managed to get it working with this version. I tried upgrading to a version > 2 but it did not fix the issue.
Regarding the LIDAR data there is no topic /velodyne. Could both errors possibly be caused by the fact that I am only starting the driver on my host PC instead of on the core?

TiBr-IGP, 2025-06-21:
## Solution
When tracking the error, the API expected a conversion of the image format from VEC2 to proto, which was not defined in the installed API version (v.4.1.1). This conflicted with the latest main branch of the repository, which requires v.5.0.0.
 
The error was resolved by manually installing API version v.5.0.0 via PIP and additionally upgrading to the latest Numpy version.

closed 2025-06-21 by TiBr-IGP