rai-opensource/spot_ros2 #694: Failed to compute the odometry pose — SLAM Toolbox incompatible TF layout with Spot Velodyne

PengzhaoWu, 2025-07-03:
Hi Boston Dynamics team,

I’m encountering an issue while integrating slam_toolbox with Spot using the spot_ros2 driver. The SLAM node repeatedly logs:
**[WARN] [slam_toolbox]: Failed to compute odom pose**
Despite publishing /scan data correctly from the Velodyne LiDAR via a custom lidar_to_scan.py node, SLAM is unable to establish the transform chain needed to compute the pose.

**System Setup:**

Robot: Boston Dynamics Spot with EAP (Velodyne)
slam_toolbox: online_async mode
Spot ROS 2 Driver: latest spot_ros2 Docker image
Config: tf_root: odom, preferred_odom_frame: odom

**What I’ve Tried:**

Set base_frame: body, odom_frame: odom, and map_frame: map in SLAM config
Published /scan from Velodyne using height filtering

**Observation:**
The TF provided by Spot’s wrapper places the LiDAR frame directly under odom by default:
odom → sensor_origin_velodyne-point-cloud
This breaks the chain expected by slam_toolbox, which assumes:
map → odom → base_frame → scan_frame

**Question:**
Is there a way to override the default TF publishing in the Spot wrapper (e.g., prevent odom → sensor_origin_velodyne-point-cloud)?
Or can this be patched in a future update to align with standard SLAM TF expectations?

PengzhaoWu, 2025-07-09:
Hi @khughes-bdai,
I hope you're doing well. I know you’re an experienced robotic software engineer—would you be able to take a look at this issue when you have a moment? I'd really appreciate your insight. Thanks in advance!

khughes-bdai, 2025-07-15:
Hi @PengzhaoWu , 
Unfortunately the point clouds come out of the Spot SDK with the parent frame as `odom`. So it is hard to change this default behavior without reaching out to them -- we are not Boston Dynamics :) 
One thing you can try to get around this is creating a node that subscribes to the point cloud topic, transforms it into a more appropriate frame, then re-publishes it on another topic. This function [`do_transform_cloud`](https://github.com/ros2/geometry2/blob/1d950de1fc28bdf882482beae56b2ad542ac9cee/tf2_sensor_msgs/tf2_sensor_msgs/tf2_sensor_msgs.py#L92) in `tf2_sensor_msgs` could be useful. Then you could use this new topic in slam toolbox?