rai-opensource/spot_ros2 #702: Body Cam's image aquisition speed control

Carson-Bay, 2025-07-16:
Currently spot_ros2 supplies images from the cams at around 3hz which is well below the expected speed from the API (I have achieved 10hz regularly and when pushing it can achieve close to 15hz). 

I understand this is computationally intensive and most of the time is not required to serve images that quickly, so providing the option and control over image rate would be ideal, specially when trying to create a video feed out of the images.

Curious on thoughts

Carson-Bay, 2025-07-17:
Amazing thats great info! 

On the same lines, is there an easy way to turn off all camera publishing to reduce processing load. I tried to set cameras_used to empty but that failed and all cameras started publishing

khughes-bdai, 2025-07-17:
Yep, you can launch the driver with the flag `launch_image_publishers:=False` and no images/point clouds will be published

khughes-bdai, 2025-07-23:
To follow up on this, I put up a small PR [here](https://github.com/bdaiinstitute/spot_ros2/pull/709) to let you manually lower these rates if needed!