rai-opensource/spot_ros2 #726: Arm velocity commands appear delayed and buffered (executing old goals minutes later)

kabirkedia, 2025-08-27:
**Description**:
I'm encountering unexpected behavior when sending velocity commands to the robot arm using the wrapper (referenced in #704). It seems like the robot is buffering and executing outdated goals—sometimes even several minutes after the commands were initially sent. This is not the intended behavior, as I need real-time control.

Here are some relevant log messages:

```
[spot_ros2-1] [INFO] [1756304302.990875882] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.008714761] [spot1.base.spot.spot_ros2]: Robot now executing goal 9390
[spot_ros2-1] [INFO] [1756304303.062938034] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.064187126] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.082767396] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.108661239] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.176619149] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.180950263] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.218769208] [spot1.base.spot.spot_ros2]: Robot now executing goal 9393
[spot_ros2-1] [INFO] [1756304303.265352024] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.269433224] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.297176255] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.370753319] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.372106934] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.379058101] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.411472955] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.459410410] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.464381944] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.473103539] [spot1.base.spot.spot_ros2]: Robot now executing goal 9397
[spot_ros2-1] [INFO] [1756304303.551965834] [spot1.base.spot.spot_ros2]: Stop attempt due to cancellation: Success
[spot_ros2-1] [INFO] [1756304303.569448963] [spot1.base.spot.spot_ros2]: Robot now executing goal 9399
[spot_ros2-1] [INFO] [1756304303.642794419] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.655507700] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.691466480] [spot1.base.spot.spot_ros2]: Robot now executing goal 9400
[spot_ros2-1] [INFO] [1756304303.781087385] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304303.787794911] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304303.824485833] [spot1.base.spot.spot_ros2]: Robot now executing goal 9401
[spot_ros2-1] [INFO] [1756304424.339586309] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304424.341603247] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304424.371837636] [spot1.base.spot.spot_ros2]: Robot now executing goal 10475
[spot_ros2-1] [INFO] [1756304424.451361390] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304424.457165107] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304424.479887921] [spot1.base.spot.spot_ros2]: Robot now executing goal 10476
[spot_ros2-1] [INFO] [1756304536.054887743] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304536.059767705] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304536.090679054] [spot1.base.spot.spot_ros2]: Robot now executing goal 11064
[spot_ros2-1] [INFO] [1756304536.168916252] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304536.177699574] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304536.219815844] [spot1.base.spot.spot_ros2]: Robot now executing goal 11065
[spot_ros2-1] [INFO] [1756304536.307333464] [spot1.base.spot.spot_ros2]: Received goal request
[spot_ros2-1] [INFO] [1756304536.310854568] [spot1.base.spot.spot_ros2]: Canceling previous goal
[spot_ros2-1] [INFO] [1756304536.352132344] [spot1.base.spot.spot_ros2]: Robot now executing goal 11067
 
```

I'm currently sending the arm velocity commands through the wrapper, and I suspect there might be a synchronization issue with the command timestamps or goal queue.

I’ve been following this example from the SDK for reference:
🔗 [https://github.com/boston-dynamics/spot-sdk/blob/master/python/examples/arm\_wasd/arm\_wasd.py](https://github.com/boston-dynamics/spot-sdk/blob/master/python/examples/arm_wasd/arm_wasd.py)

**Questions**:

* Could this delayed execution be due to improper timestamp handling or clock sync between ROS and the Spot robot?
* Is there a way to ensure commands are dropped if they become stale instead of being queued?
* Are there best practices for canceling or preempting old arm goals to avoid this behavior?

Tagging @khughes-bdai @tcappellari-bdai for visibility.

Let me know if logs or additional details would help. Thanks!

tcappellari-bdai, 2025-08-27:
Are you connecting with the robot over wifi or ethernet? I usually find there's more of a delay between sending a command the robot executing it when using wifi vs. ethernet.

kabirkedia, 2025-08-27:
Sure.
Do you just want to look at the callback or do you want something you can compile and run on the robot

kabirkedia, 2025-09-02:
Hi @khughes-bdai sorry for the late reply.
I will add an example for wasd with arm pretty soon but I think I identified the issue.
I think the issue came from the timesync endpoints not being synced from the computer to the robot, which was really unexpected.

When I compared the cmd_vel callback (body movement) with the arm_vel callback (arm movement), I noticed a key difference:

In cmd_vel, the endpoint is passed:
[https://github.com/bdaiinstitute/spot_wrapper/blob/main/spot_wrapper/wrapper.py#L1328](https://github.com/bdaiinstitute/spot_wrapper/blob/main/spot_wrapper/wrapper.py?utm_source=chatgpt.com#L1328)

khughes-bdai, 2025-09-02:
Ahh good catch. All of the other standard `robot_command`s do this as well: https://github.com/bdaiinstitute/spot_wrapper/blob/423257ade5e22fd1164218670541dc415dac7c6c/spot_wrapper/wrapper.py#L1441

tcappellari-bdai, 2025-09-02:
> Ah, got it! That was the only difference I noticed, so I thought that might have been the issue. I also had a quick question about QoS—shouldn’t all velocity commands use Best Effort to avoid them getting queued up and instead always just honor the latest one?

Yes, they ideally should but `image_pipeline` had a bug in it at the time that prevented us from (easily) modifying the QoS. I believe the issue has since been fixed and released but we haven't had the time to get back around to it

kabirkedia, 2025-09-02:
Would you like me to make that change in the PR? I’ve been running into the buffering issue myself, and from what I’ve heard, the recommended approach is to use Best Effort with a history depth of 0.

Also, I was thinking of adding another feature—a service to lock the arm in place. Doing it directly through the SDK doesn’t seem like the best long-term approach. What are your thoughts on this?

tcappellari-bdai, 2025-09-02:
It would be super helpful if you're able to make the QoS change and open a PR!

As for locking the arm in place, I'd be ok with adding that (as a separate PR though) unless @khughes-bdai has any objections

kabirkedia, 2025-09-03:
Hey @tcappellari-bdai @khughes-bdai, I noticed one more thing. In the driver [[here](https://github.com/bdaiinstitute/spot_wrapper/blob/423257ade5e22fd1164218670541dc415dac7c6c/spot_wrapper/wrapper.py#L1312)](https://github.com/bdaiinstitute/spot_wrapper/blob/423257ade5e22fd1164218670541dc415dac7c6c/spot_wrapper/wrapper.py#L1312), the time is currently being reset with `time.time()`.

Shouldn’t this instead use the timestamp from the ROS message header? That way the driver preserves the original timing information from the message, rather than overwriting it.

I think the same consideration applies to arm velocity, joint commands, and similar messages as well.

kabirkedia, 2025-09-03:
What callbacks would you like me to update this for?

I was thinking of changing it in:

* `cmd_vel`
* `arm_velocity_commands`
* `arm_pose_commands`
* `arm_joint_commands`

Basically all of them 😅

Also, I put together a new QoS profile:

```python
# Recommended QoS for Joy/Teleop commands
JOY_TELEOP_QOS = QoSProfile(
    # Best Effort - prioritizes low latency over guaranteed delivery
    # Perfect for real-time control where latest command matters most
    reliability=ReliabilityPolicy.BEST_EFFORT,
    
    # Keep Last with small depth - only care about most recent commands
    # History of 1–5 is typical, 1 is often sufficient for teleop
    history=HistoryPolicy.KEEP_LAST,
    depth=1,  # Only keep the latest command
    
    # Volatile - don't persist commands after node restart
    # Teleop commands shouldn't be replayed from before restart
    durability=DurabilityPolicy.VOLATILE,

    # Short lifespan - commands expire quickly to avoid stale commands
    lifespan=rclpy.duration.Duration(seconds=1.0),  # Commands expire quickly
)
```

Right now, I’ve only added this to the callbacks for `arm_joint_commands`, `arm_velocity_commands`, and `cmd_vel`.

Do you want me to apply this profile anywhere else as well?

kabirkedia, 2025-09-03:
Edit after looking into this I understand only teleop topics have explicit timestamp handling. Is that by design?

kabirkedia, 2025-09-03:
In case of teleop, from our experience it's important that the robot knows the timestamp directly coming from the joystick(or some other controller).
We had an instance where the robot lost connection to the joystick for like 5 seconds and ROS buffered the commands to reliable QoS.
It didn't stop moving.
If we pass the timestamp coming from the joystick directly, the api will know that this command had expired. That's why I was advocating for time stamping the commands through the header of the ros message. And that's also why I changed the publisher from twist to twist stamped. 
Do you think that's not really relevant here?

khughes-bdai, 2025-09-04:
If you change the QOS settings to the profile you specified above, does that resolve the buffering behavior? 
It does make sense in your use case to have the option to specify the header in this topic, but I would like to avoid changing the message type of `cmd_vel` as that would be a breaking change of a commonly used topic. So my opinion is if the QOS change fixes the problem then that would be the best solution to merge in since that keeps the user interface the same.