rai-opensource/spot_ros2 #727: Spot CAM payload images not showing up on ros2 topic list

kennycaldas, 2025-08-28:
I am trying to use the Spot CAM with the spot_driver, but even after setting the parameter `initialize_spot_cam: True`, there are no topics available when running the driver. Checking on ros2 service list, it is possible to see some parameters like `/spot/get_ptz_position`, and `/spot/set_ptz_position` related to the payload. In fact, setting the ptz position moves the Spot CAM, but I could not find anything related to the cameras images. Do you have any idea what might be happening?

khughes-bdai, 2025-08-28:
We disabled the image publishing on Spot cam because we were unable to get images through the BD SDK -- see https://github.com/bdaiinstitute/spot_wrapper/pull/126 and https://github.com/bdaiinstitute/spot_wrapper/issues/127. You can try re-enabling the line in the wrapper that was commented out in that PR in case anything has changed, at the time we were stuck on errors like this:
```
[h264 @ 0x787898001340] non-existing PPS 0 referenced
[h264 @ 0x787898001340] decode_slice_header error
[h264 @ 0x787898001340] no frame!
```
@mhidalgo-bdai did we ever hear back from support about ways to get around this issue? My understanding was that streaming from the spot cam was working through the web browser interface but not through the Python SDK.

mhidalgo-bdai, 2025-08-29:
@khughes-bdai I did not, but there's been some traffic on https://github.com/bdaiinstitute/spot_wrapper/issues/127 since. 

@kennycaldas would you be willing to revert https://github.com/bdaiinstitute/spot_wrapper/pull/126 and apply the change that https://github.com/bdaiinstitute/spot_wrapper/issues/127#issuecomment-2426991436 suggested? If it works for you, then maybe it is time we re-enable.